androidthings/sample-uartloopback #1: Android things using UART on Intel Edison

avirup171, 2017-02-06:
Hi,
The sample was tried using an Intel Edison but it seems not to work at all. 

[image]

The data you are seeing is not sent by me. 

The code for transferUartData() is mentioned below

```
private void transferUartData() {
        if (mLoopbackDevice != null) {
            // Loop until there is no more data in the RX buffer.
            try {
                //Log.e(TAG, "Reading1");
                byte[] buffer = new byte[CHUNK_SIZE];
                mLoopbackDevice.write(buffer, 45);
                int read;
                while ((read = mLoopbackDevice.read(buffer, buffer.length))!=0) {
                    mLoopbackDevice.write(buffer, read);
                    Log.e(TAG, String.valueOf(read));
                }
            } catch (IOException e) {
                Log.w(TAG, "Unable to transfer data over UART", e);
            }
        }
    }
```
And I am not sending anything through the cable. 

If possible, guide me to implement the same thing on an Intel Edison on an Arduino breakout board. Do I need to do any extra steps for the UART or anything will be helpful.

devunwired, 2017-02-06:
Can you provide some more details about your setup? Have you connected to your Edison the same way as the [schematics in the README](https://github.com/androidthings/sample-uartloopback#schematics), or are you doing something different. Perhaps RX/TX are crossed, or the [baud rate on your computer](https://github.com/androidthings/sample-uartloopback/blob/master/app/src/main/java/com/example/androidthings/loopback/LoopbackActivity.java#L40) is set incorrectly?

avirup171, 2017-02-07:
I did the circuitry as mentioned in the image. RX-TX, TX-RX, GND-GND. Baud rate is set to 9600
When I run the code, I keep on getting some random data. I am using the Arduino IDE serial port terminal. 
in the terminal, i get some random characters as shown below.

[image]

Is there any workaround that can be done to solve the problem, my main target is to get analog sensor data by UART

devunwired, 2017-02-07:
I just realized you modified the sample code in your comment above. What is the purpose of this line?
```
mLoopbackDevice.write(buffer, 45);
```
This will cause 45 bytes of null data to be written out every time `transferUartData()` is called regardless of anything being read. That probably accounts for much of your garbage data, but that alone doesn't explain why the callback is triggering repeatedly. That sounds more like a hardware problem (like the RX line of the Edison is being held low or something like that).

avirup171, 2017-02-07:
I added that line to check whether data can be written or not. Apparently the Edison is only connected to the USB port as mentioned in the android things documentation. So if it's a hardware problem, then is it because of low power or so? Should I put it on external power supply?

Currently, I am not able to neither read nor write data using UART.

devunwired, 2017-02-08:
You could also try shorting RX/TX together on the Edison and not have the USB-TTL cable connected to eliminate any potential issues caused by that for the moment. You could modify the sample code to periodically write out a fixed string and verify that you are able to read the same string back.

avirup171, 2017-02-09:
Yesterday, wehn we were testing using another board, we did get some random values much like some random values. I'll give a shot using the method above and see if it works or not. I wanted to know one more thing, is it necessary to set anything on the Intel Edison as the UART? beside running the sample?

devunwired, 2017-02-09:
The code to enable & configure the UART is found in the `openUart()` method. There are no additional steps (in hardware or software) for the Edison. I don't know which USB-TTL cable you were using, but the IOREF voltage (set by J9) would also need to match your cable (5V or 3.3V).