androidthings/sample-uartloopback #3: Cannot receive data on RPI3 (cannot write to communication pipe)

bsodam, 2017-06-14:
Hello, I tried this sample and it does not work for me. 
I tried it on RPI3 with Android Things developer preview 4.
I turned off the UART debugging mode of RP3 by changing "console=serial0,115200" to "console=tty0" in cmdline.txt, and adding "dtoverlay=pi3-disable-bt" and removing "enabled_uart=1" & "core_freq=400" in config.txt file.

When I run this sample, I typed some characters in terminal program but I cannot get any echo back.

From the logs, I guess there are some problems opening some files related to communication pipe.
Can you give me any ideas to solve this situation?
Thanks

Below is the logs from logcat.

`06-14 00:34:15.117 7959-7959/? E/cutils-trace: Error opening trace file: No such file or directory (2)
                                               
                                               [ 06-14 00:34:15.145   156:  156 I/         ]
                                               free_cache(3075) avail 2575781888
06-14 00:34:15.268 7959-7959/? W/art: Unknown instruction set features for ARM CPU variant (generic) using conservative defaults
06-14 00:34:15.283 7959-7959/? D/ICU: No timezone override file found: /data/misc/zoneinfo/current/icu/icu_tzdata.dat
                                      
                                      [ 06-14 00:34:15.295   156:  156 I/         ]
                                      free_cache(28380) avail 2575777792
06-14 00:34:15.358 7959-7959/? E/memtrack: Couldn't load memtrack module (No such file or directory)
06-14 00:34:15.358 7959-7959/? E/android.os.Debug: failed to load memtrack module: -2
06-14 00:34:15.361 7959-7959/? I/Radio-JNI: register_android_hardware_Radio DONE
06-14 00:34:15.397 7959-7959/? D/AndroidRuntime: Calling main entry com.android.commands.wm.Wm
06-14 00:34:15.402 7959-7959/? D/AndroidRuntime: Shutting down VM
06-14 00:34:15.756 156-156/? I/SELinux: SELinux: Loaded file_contexts contexts from /file_contexts.bin.
06-14 00:34:15.758 410-435/system_process I/PackageManager.DexOptimizer: Running dexopt (dex2oat) on: /data/app/vmdl1792012311.tmp/base.apk pkg=com.example.androidthings.loopback isa=arm vmSafeMode=false debuggable=true target-filter=interpret-only oatDir = /data/app/vmdl1792012311.tmp/oat sharedLibraries=/system/framework/com.google.android.things.jar
06-14 00:34:15.891 8038-8038/? W/dex2oat: Unknown instruction set features for ARM CPU variant (generic) using conservative defaults
06-14 00:34:15.892 8038-8038/? I/dex2oat: /system/bin/dex2oat --compiler-filter=interpret-only --debuggable
06-14 00:34:15.892 8038-8038/? E/cutils-trace: Error opening trace file: No such file or directory (2)
06-14 00:34:16.066 8038-8038/? I/dex2oat: dex2oat took 175.564ms (threads: 4) arena alloc=9KB (9800B) java alloc=111KB (114104B) native alloc=1199KB (1228760B) free=1360KB (1392680B)
                                          
                                          --------- beginning of system
06-14 00:34:16.281 410-435/system_process E/PackageManager: Backup Manager not found!
06-14 00:34:16.303 410-435/system_process I/art: Starting a blocking GC Explicit
06-14 00:34:16.429 410-435/system_process I/art: Explicit concurrent mark sweep GC freed 24161(1639KB) AllocSpace objects, 26(868KB) LOS objects, 33% free, 6MB/9MB, paused 2.442ms total 125.404ms
                                                 
                                                 [ 06-14 00:34:16.430   156:  156 E/         ]
                                                 Couldn't opendir /data/app/vmdl1792012311.tmp: No such file or directory
06-14 00:34:16.434 410-988/system_process I/System.out: Success
06-14 00:34:16.489 410-449/system_process I/InputReader: Reconfiguring input devices.  changes=0x00000010
06-14 00:34:16.557 410-423/system_process W/Searchables: No global search activity found
06-14 00:34:16.593 809-8050/com.google.android.gms I/PeopleContactsSync: triggerPendingContactsCleanup: no accounts
06-14 00:34:16.600 809-8050/com.google.android.gms D/Wear_Controller: Received broadcast action=android.intent.action.PACKAGE_ADDED and uri=com.example.androidthings.loopback
06-14 00:34:16.696 809-911/com.google.android.gms I/Icing: Usage reports 0 indexed 0 rejected 0 imm upload false
06-14 00:34:16.706 809-911/com.google.android.gms I/Icing: Usage reports 0 indexed 0 rejected 0 imm upload false
06-14 00:34:16.712 809-8051/com.google.android.gms W/IcingInternalCorpora: getNumBytesRead when not calculated.
06-14 00:34:17.715 809-7908/com.google.android.gms I/Icing: Indexing 552A18E96A12CBAD55F637AC8023C0E0F78C68F3 from com.google.android.gms
06-14 00:34:17.794 809-7908/com.google.android.gms D/Icing: Loaded CLD2 Version V2.0 - 20141016
06-14 00:34:17.804 809-7908/com.google.android.gms I/Icing: Indexing done 552A18E96A12CBAD55F637AC8023C0E0F78C68F3
06-14 00:34:18.209 8054-8054/? D/AndroidRuntime: >>>>>> START com.android.internal.os.RuntimeInit uid 2000 <<<<<<
06-14 00:34:18.219 8054-8054/? D/AndroidRuntime: CheckJNI is OFF
06-14 00:34:18.220 8054-8054/? E/cutils-trace: Error opening trace file: No such file or directory (2)
06-14 00:34:18.356 8054-8054/? W/art: Unknown instruction set features for ARM CPU variant (generic) using conservative defaults
06-14 00:34:18.371 8054-8054/? D/ICU: No timezone override file found: /data/misc/zoneinfo/current/icu/icu_tzdata.dat
06-14 00:34:18.447 8054-8054/? E/memtrack: Couldn't load memtrack module (No such file or directory)
06-14 00:34:18.447 8054-8054/? E/android.os.Debug: failed to load memtrack module: -2
06-14 00:34:18.450 8054-8054/? I/Radio-JNI: register_android_hardware_Radio DONE
06-14 00:34:18.485 8054-8054/? D/AndroidRuntime: Calling main entry com.android.commands.am.Am
06-14 00:34:18.523 410-649/system_process I/ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.example.androidthings.loopback/.LoopbackActivity} from uid 2000 on display 0
06-14 00:34:18.548 8054-8054/? D/AndroidRuntime: Shutting down VM
06-14 00:34:18.565 410-422/system_process I/ActivityManager: Start proc 8064:com.example.androidthings.loopback/u0a29 for activity com.example.androidthings.loopback/.LoopbackActivity
06-14 00:34:18.576 8064-8064/? I/art: Late-enabling -Xcheck:jni
06-14 00:34:18.578 8064-8064/? W/art: Unknown instruction set features for ARM CPU variant (generic) using conservative defaults
06-14 00:34:18.761 8064-8064/com.example.androidthings.loopback W/System: ClassLoader referenced unknown path: /data/app/com.example.androidthings.loopback-1/lib/arm
06-14 00:34:18.893 8064-8064/com.example.androidthings.loopback D/LoopbackActivity: Loopback Created
06-14 00:34:18.887 163-163/? I/peripheralman: type=1400 audit(0.0:36): avc: denied { dac_override } for capability=1 scontext=u:r:peripheralman:s0 tcontext=u:r:peripheralman:s0 tclass=capability permissive=1
06-14 00:34:18.887 163-163/? I/peripheralman: type=1400 audit(0.0:37): avc: denied { read write } for name="ttyAMA0" dev="tmpfs" ino=9297 scontext=u:r:peripheralman:s0 tcontext=u:object_r:device:s0 tclass=chr_file permissive=1
06-14 00:34:18.887 163-163/? I/peripheralman: type=1400 audit(0.0:38): avc: denied { open } for path="/dev/ttyAMA0" dev="tmpfs" ino=9297 scontext=u:r:peripheralman:s0 tcontext=u:object_r:device:s0 tclass=chr_file permissive=1
06-14 00:34:18.887 163-163/? I/peripheralman: type=1400 audit(0.0:39): avc: denied { ioctl } for path="/dev/ttyAMA0" dev="tmpfs" ino=9297 ioctlcmd=5401 scontext=u:r:peripheralman:s0 tcontext=u:object_r:device:s0 tclass=chr_file permissive=1
06-14 00:34:19.097 410-649/system_process W/art: Long monitor contention with owner android.display (431) at void com.android.server.wm.WindowAnimator$1.doFrame(long)(WindowAnimator.java:138) waiters=0 in float com.android.server.wm.WindowManagerService.getCurrentAnimatorScale() for 149ms
06-14 00:34:19.879 410-440/system_process W/art: Long monitor contention with owner android.display (431) at void com.android.server.wm.WindowAnimator$1.doFrame(long)(WindowAnimator.java:138) waiters=0 in int com.android.server.wm.WindowManagerService.addWindow(com.android.server.wm.Session, android.view.IWindow, int, android.view.WindowManager$LayoutParams, int, int, android.graphics.Rect, android.graphics.Rect, android.graphics.Rect, android.view.InputChannel) for 342ms
[... cut ...] eAnimator.finishExit:565 com.android.server.wm.WindowStateAnimator.stepAnimationLocked:491 com.android.server.wm.WindowAnimator.updateWindowsLocked:303 com.android.server.wm.WindowAnimator.animateLocked:704 
06-14 00:34:21.547 410-480/system_process E/TaskPersister: File error accessing recents directory (directory doesn't exist?).
06-14 00:35:09.632 163-270/? E/peripheralman: [0614/003509:ERROR:uart_driver_sysfs.cc(388)] Can't write to the communication pipe. Something is not right with client`

tyharness, 2017-06-15:
Haven't had chance to check but I think
enabled_uart=1 should be left in.  Will check later and report back.  Could it be a typo miss print on the web page??
https://developer.android.com/things/hardware/raspberrypi.html#disabling_the_console

bsodam, 2017-06-16:
Thank you for your comment.

I was not familiar with UART communication, so I misunderstood the sample.
When I typed a letter in terminal program, I can see the letter I typed.  I thought additional letter should be echo back from UART serial. 

But now I found out that if I can see the letter I type, It means it works.
I hope that someone has same issue with me could see this.

closed 2017-06-16 by bsodam