androidthings/sample-uartloopback #4: UART  divides message when writing it

mahermeg17, 2017-11-28:
Hi,
I'm using this sample code with a PICO PI IMX6UL.
The problem is when I send a String like "abcdefghijklmno". I receive it on the other side divided, it means not as a single string,  like "abc" then "de" then "fghijk" then "lmno".
And messages are received into random way, not the same number of parts, and not the same length of each part.
My UART config is as following:
BAUD_RATE = 115200;
DATA_BITS = 8;
STOP_BITS = 1;
CHUNK_SIZE = 256;

On the other side, I use UsbSerial library on an android phone (https://github.com/felHR85/UsbSerial)

I need to read data in one piece. Any help please!
Thanks.

Fleker, 2017-11-28:
This might be a bug in the UsbSerial library.

But in general, you should expect UART data to come in intermittently. Although it may stream continually, the interrupts may be fired at any point. You should buffer the data and determine the stop point on your own, then process it.

Generally, embedded engineers will do several different steps to properly segment the data:

* Use a fixed size, ie. assume every packet is 3 bytes
* Send a fixed-size header which contains information about the data such as the total length and perhaps error correction information
* Send the length of the packet as the first byte, allowing the receiver to count that many bytes (with a max of 255 characters)
* Send a stop character at the end of your data transmission, ie. `0x00`. This is more useful if you're sending strings. Many systems append a null character at the end of strings internally to denote that you have reached the end.

mahermeg17, 2017-11-29:
Thank you for the reply.
it seems to be caused by the UsbSerial lib. I found a similar bug on the project issues
https://github.com/felHR85/UsbSerial/issues/92

Bests,

closed 2017-11-29 by mahermeg17

mahermeg17, 2017-11-29:
I didn't have this problem when using https://github.com/mik3y/usb-serial-for-android