androidthings/sample-uartloopback #5: The length of each receiving of UART

Krguang, 2017-12-05:
I'm going to use android things to implement the modbus protocol，But I've found that when android things receives a piece of data, it's automatically divided into many segments. Is there any way to get it to be done like a single chip?
[image]
[image]

Fleker, 2017-12-05:
This seems similar to #4. I'll copy my response there.

In general, you should expect UART data to come in intermittently. Although it may stream continually, the interrupts may be fired at any point. You should buffer the data and determine the stop point on your own, then process it.

Generally, embedded engineers will do several different steps to properly segment the data:

Use a fixed size, ie. assume every packet is 3 bytes
Send a fixed-size header which contains information about the data such as the total length and perhaps error correction information
Send the length of the packet as the first byte, allowing the receiver to count that many bytes (with a max of 255 characters)
Send a stop character at the end of your data transmission, ie. 0x00. This is more useful if you're sending strings. Many systems append a null character at the end of strings internally to denote that you have reached the end.

Krguang, 2017-12-05:
Thanks for your answer!
The modbus protocol does not have a starting character , a stop character , or a character that represents length , We can only be judged by time interval.
I tried to use TimerTask to make judgments, but because I was a beginner, there were occasional array overflows

Fleker, 2017-12-05:
Maybe you can use an `ArrayList` and convert that to a buffer after?

Is there any sort of expectation you can make about the data? When I've used Modbus, there have been certain types of expected payloads that made it easier to implement.

Krguang, 2017-12-05:
Thanks again,My mind is still stuck on C. I'll try to use arraylist. 
I don't quite understand what you mean by "expected payloads".
Is it the java library that implements the modbus？

Fleker, 2017-12-05:
If I'm talking from Android Things to a Modbus peripheral, or vice-versa, there should be some established protocol of the types of messages being sent. You should be able to use that protocol in order to find when to do the cut-off.

Krguang, 2017-12-05:
I think I see what you mean，like GPS protocol，
I found out $GPGGA，I began to process the data.
but the modbus protocol have no specific start characters.
I can only judge by time，So I need to get a full frame to process.

Fleker, 2017-12-05:
If there's no way to judge with start/end characters, you may be able to log the timestamp of each UART event and split it that way.

Krguang, 2017-12-06:
OK, I will try to do this and thanks for your kind help