androidthings/sample-uartloopback #6: UART callback not triggered in Raspberry pi running dp6.0?

rahulraj64, 2017-12-08:
UART callbacks not triggering on Raspberry Pi, running Android  Things DP 6.0. Do we need edit the config or cmdline files to change serial & bluetooth parameters, as we did in previous developer previews?  I use a USB serial software called CoolTerm for Mac to check the connection. I also checked with a UART device ( Rs232 converted to UART) too. It is not working. The connections are correct. When the Rapsberry pi boots, I am able to read  console log from the Usb serial software. 
log: 
_............
[    3.591832] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[    3.591837] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.595054] hub 1-1:1.0: USB hub found
[    3.595113] hub 1-1:1.0: 5 ports detected
[    3.606690] mmc1: new high speed SDIO card at address 0001
[    3.622230] Key type dns_resolver registered
[    3.627361] Registering SWP/SWPB emulation handler
[    3.632860] registered taskstats version 1
[    3.640041] runtimepinconfig: UART0 stopping at 2 pins out of 3
[    3.646069] runtimepinconfig: UART0 stopping at 2 pins out of 3
rpi3:/ $_ 

But unable to get the UART callback when try to give input either from the serial software or from UART device.

fluxxion82, 2018-02-11:
I'm having issues as well. Running DP 0.6.1. Using Silicon Laboratories driver and USB to UART bridge controller, and using the Serial app for mac os. I can see things printing on the console when the rpi is booting, but when I run the sample loopback app, the Serial app appears to freeze (unfreezes when I kill the loopback app). Don't see any errors of any kind and I don't see any evidence that the callbacks are triggering from the device or Serial app.

Fleker, 2018-02-12:
Are you able to read any UART data on the Android Things device?

fluxxion82, 2018-02-15:
So I basically copied the code from the `transferUartData` method to the end of `openUart`method and didn't get anything from the `read` call.

Fleker, 2018-02-15:
If you can't read anything either asynchronously or synchronously, that leads me to believe the issue is not in the callback but in the peripheral's hardware or in the wired connection.

The RPi prints out its boot process to serial, so it's not a Pi issue. Can you attach a logic analyzer on the same UART pins and see if there's a data connection issue?

fluxxion82, 2018-02-16:
What is there to read when I after the uart opens? The app is supposed to read the input from the serial, but when there is no input, what else would it read?
There's a logic analyzer at work that I can use tomorrow. I'll see what I can find out from that..

Fleker, 2018-02-16:
When the UART first opens, there will not be any data sent on that bus. However, if the device is sending data and you're not seeing it, then there could be some sort of communication issue.

fluxxion82, 2018-02-16:
Is there another way I can send data if the serial app is frozen? When I modified the sample app to try and read soon after I open uart, I wasn't explicitly sending any data.

Fleker, 2018-02-16:
Does it freeze as in not responding? What are you expecting would happen? It seems like Serial performs exactly as expected. If no UART data is being sent, then it would not be doing anything.

fluxxion82, 2018-02-17:
Yea it freezes and I can't input anything nor do I see anything show up. It's definitely not as expected. As soon as I kill the app, I can type in serial again.

fluxxion82, 2018-02-17:
Ok so I hooked up a logic analyzer and ran the app, and I can see data getting transmitted and received. I'm running picocom on linux and I think I'm seeing the echo through picocom. I guess there's an issue with Serial on mac os?

fluxxion82, 2018-02-17:
I got picocom running on my mac and got the sample app to run and it looks like it's working. So I guess there's just some issue with the Serial app. Thanks for your help @Fleker

closed 2018-02-20 by Fleker