androidthings/sample-uartloopback #8: UART Cant open

Ragulmurugan, 2018-02-12:
E/Exception: UART cant opencom.google.android.things.pio.PioException: android.os.ServiceSpecificException: UART0 is already in use (code 16)

Fleker, 2018-02-12:
Seems like you've opened it in another activity without closing it. See [this Stackoverflow answer](https://stackoverflow.com/questions/48437428/android-os-servicespecificexception-gpio2-io02-is-already-in-use-code-16) to learn more.

Ragulmurugan, 2018-02-15:
But I'm closing UART before passing the Intent in every activity and this problem occurs randomly if it occurs always I can consider I forget to Close somewhere even I had the main thread.sleep before passing the Intent but nothing solved this problem.

Fleker, 2018-02-15:
Where are you closing it? In the activity's `onDestroy` method? Do you need to architect your application with multiple activities?

Ragulmurugan, 2018-02-16:
I'm closing it before passing the intent like 

                try {
                    closeUart();
                } catch (IOException e) {
                    Log.e(TAG, "Error closing UART device:", e);
                }
                Intent i = new Intent(MainActivity.this,MainActivity2.class);
                String strName = "j";
                i.putExtra(Intent.EXTRA_TEXT, strName);
                startActivity(i);
                overridePendingTransition(android.R.anim.fade_in, android.R.anim.fade_out);finish();
                finish();

I'm not closing it in onDestroy method.

Fleker, 2018-02-16:
Can you add a type of asynchronous wait before starting your activity, like using a Handler:

```Java
try {
    closeUart();
} catch (IOException e) {
    Log.e(TAG, "Error closing UART device:", e);
}
h = new Handler(Looper.getMainLooper()) {
    @Override
    public void handleMessage(int what) {
        Intent i = new Intent(MainActivity.this,MainActivity2.class);
        String strName = "j";
        i.putExtra(Intent.EXTRA_TEXT, strName);
        startActivity(i);
        overridePendingTransition(android.R.anim.fade_in, android.R.anim.fade_out);finish();
        finish();
    }
};
h.sendEmptyMessageDelayed(0, 50);
```

(this code wasn't tested, so it may not be completely valid)

Ragulmurugan, 2018-02-18:
I have tried using main thread timer before passing the intent. Like
Thread.sleep(1000).
This time I'll try your idea let me use handler  .
Thanks dude