androidthings/sample-uartloopback #10: Random code problem

fail572872916, 2018-11-05:
I want to input the content I input in aruino at Log, but there is a garbled problem and how to solve it.
[image]
[image]

Fleker, 2018-11-06:
Can you print out the character codes of each character? My guess, from looking at this, is that you're reading in a ton of characters at a time, more than the sending device is sending. As such, it is going to read in one character and leave the rest as nulls (character code '0').

Keep in mind the data types you're passing back and forth. If your Arduino is sending numbers, you should be looking at the numbers coming back and not the associated characters.

fail572872916, 2018-11-06:
I have solved this problem.
[image]

closed 2018-11-06 by fail572872916