androidthings/sample-uartloopback #14: App get crashed when i use PeripheralManager.

imbichitra, 2019-11-27:
Hi ,
The app is getting chrashed when i use `PeripheralManager.getInstance();`
below is error given.
```
Caused by: java.lang.RuntimeException: Stub!
        at com.google.android.things.pio.PeripheralManager.getInstance(PeripheralManager.java:21)
        at com.asiczen.usb1.UartActivity.onCreate(UartActivity.java:43)
        at android.app.Activity.performCreate(Activity.java:6858)

Fleker, 2019-11-27:
You should make sure that `<uses-library android:name="com.google.android.things"/>` is in your project manifest and that your device is infact running Android Things.

gamenic-babin, 2020-10-07:
Yes my device android 10 and sdk tool is 26 .. added android things dependency permission on manifest and still getting this error

gamenic-babin, 2020-10-07:
I added above uses to manifest it gives build error , install failed missing shared library

ajaykeshri, 2021-12-17:
is there any solution for this? I am also getting same issue.
Please help