androidthings/sample-uartloopback #15: App got crashed when launch into physical device

UnisonControls, 2020-06-12:
Dear all,

We are trying to communicate android device to dsPIC series MCU using PL2303 USB to Uart converter.
We check all hardware on windows and linux PC and it is working ok.

We just download this sample source code for JAVA language and open it in linux pc where android studio 4.0 with gradle version 6.1 are available.
We connect physical device (MOTO G5 with Android 8, Redmi Note 9 pro with Android 10) on PC USB and run app from android studio both both device individually. 
But both time we got same error as mentioned below.

```
Installation did not succeed.
The application could not be installed: INSTALL_FAILED_MISSING_SHARED_LIBRARY

List of apks:
[0] '/home/user/AndroidStudioProjects/UARTLoopback/app/build/outputs/apk/debug/app-debug.apk'
Installation failed due to: 'null'
```

To resolve this issue we tried some google search where most common solution we found is as below link.
[https://github.com/androidthings/new-project-template/issues/1]

After adding `android:required="false"` this line in AndroidManifest.xml file we are able to run app on device. 
But whenever we open app on device app got forced stop and showing below error logcat.
[consollog.txt](https://github.com/androidthings/sample-uartloopback/files/4769248/consollog.txt)

We would link to overcome with this problem ASAP.

Waiting for your quick reply.

Thanks and Best Regards,
Jehan Patel

Fleker, 2020-06-12:
> MOTO G5 with Android 8, Redmi Note 9 pro with Android 10

These devices are not running Android Things, and do not have the necessary libraries to work with UART. Those devices will be unable to run this sample.

UnisonControls, 2020-06-13:
Can you please tell me which devices can support android things?
We want to make universal app to send and receive data using OTG.
Please find attachment for getting clear concept of our project need.
Also let me know the error present due to android things activity not support?
Is there any solution available to run this project in any device? (our targeted sdk is lolipop version and all higher version of android) 
[image]

jdkoren, 2020-06-13:
This is the only officially supported hardware for Android Things is listed [here](https://developer.android.com/things/hardware).

Do note that Android Things has transitioned to a platform for OEM partners ([source](https://android-developers.googleblog.com/2019/02/an-update-on-android-things.html)), and that the hardware listed in the page above might no longer be available.

UnisonControls, 2020-06-15:
Hi,

Thanks for the clarification and supported links. 
But right now we are not using any RTOS system as mentioned on your comment links.

**We want to make Universal Android device app which can send UART commands to any microcontroller using OTG or data cable or charging cable. 
Please find attached block diagram on previous comment.** 

Is it possible to send UART command from android app?

Feel free to ask if you have any query.

jdkoren, 2020-06-15:
The peripheral I/O APIs, such as for UART, are part of the [Things Support Library][2]. This library does not exist on any standard Android device, it only exists on the supported hardware I linked to in my [previous comment][1].

The Things Support Library is not third-party library code that you build into your app. It is part of the operating system image and it is linked at runtime, the same as the regular Android APIs are linked at runtime, therefore it is not available on any other hardware. That is why you get the `java.lang.NoClassDefFoundError` when you try to use these APIs on a phone.

If you are using a USB cable, perhaps you can try the [USB APIs][3] to send data.

[1]: https://github.com/androidthings/sample-uartloopback/issues/15#issuecomment-643688822
[2]: https://developer.android.com/things/get-started#sdk
[3]: https://developer.android.com/guide/topics/connectivity/usb

UnisonControls, 2020-06-16:
Hi,

Thanks for giving me correct way.

We can send or receive data using USB host API to other peripherals.

[](https://github.com/mik3y/usb-serial-for-android)

this link can help us more .

Thanks for your support we can close this issue.

closed 2020-06-16 by UnisonControls